ArduPilot/ardupilot_wiki #4823: Dead Reckoning Failsafe

billbecher, 2022-12-20:
[Dead Reckoning Failsafe](https://ardupilot.org/copter/docs/deadreckoning-failsafe.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I do not see any of the RC 3 options in Parameter list. Are the instructions on this page still applicable? Or do I need to use the script for dead reckoning?

Hwurzburg, 2022-12-20:
what RC3_OPTION are you talking about, this page never mentions that...it does talk about setting up an RC9_OPTION to kill GPS for testing....what are you referring to?

billbecher, 2022-12-20:
I see EK3_DRAG_BCOEF_X as needed to set up the winspeed 
I am new to ArduPilot and can't figure out how to change failsafe to dead reckoning. 
I have ver 4.3.1 installed

billbecher, 2022-12-20:
I meant EK3 but basically need assistance setting up dead reckoning failsafe

billbecher, 2022-12-20:
I will try to get failsafe mode to dead reckoning tomorrow and if still problems will post more accurate questions and screen shots from Mission Planner