ArduPilot/ardupilot_wiki #4826: Cameras and Gimbals

billbecher, 2022-12-21:
[Cameras and Gimbals](https://ardupilot.org/copter/docs/common-cameras-and-gimbals.html) -

 Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Link goes to page where it says needs AP4.3.1 or higher not 4.4. I thought 4.3.1 is the latest stable version.

davidbuzz, 2022-12-21:
ok, so https://ardupilot.org/copter/docs/common-cameras-and-gimbals.html says "only supported in AP 4.4 and higher" and https://ardupilot.org/copter/docs/common-siyi-zr10-gimbal.html#common-siyi-zr10-gimbal says "available in ArduPilot 4.3.1 (and higher)" - i guess one of these is a typo.

rmackay9, 2022-12-21:
Thanks for this.  The landing page is incorrect.  I'll fix it.

rmackay9, 2022-12-22:
Here's the PR to fix this https://github.com/ArduPilot/ardupilot_wiki/pull/4828

rmackay9, 2022-12-22:
This is fixed now, thanks again for the report.

closed 2022-12-22 by rmackay9