ArduPilot/ardupilot_wiki #4947: Setting up the waf Build Environment on Windows using Cygwin

AlexL922, 2023-02-18:
[Setting up the waf Build Environment on Windows using Cygwin](https://ardupilot.org/dev/docs/building-setup-windows-cygwin.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

When installing the relevant packages following step 2 of "Setting up the waf Build Environment on Windows using Cygwin", 'cygwin32-gcc-g++' is not available in the list of packages. Please can someone help with this.

Hwurzburg, 2023-02-18:
we really do not recommend using Cygwin...use WSL2

TunaLobster, 2023-02-18:
The cygwin environment install script should grab the correct package. The table on the wiki page might be out of date though.

AlexL922, 2023-02-18:
> we really do not recommend using Cygwin...use WSL2
Okay thank you I will try WSL2 instead

> The cygwin environment install script should grab the correct package. The table on the wiki page might be out of date though.
Yeah I was thinking that the page might be out of date, thanks for your help.

Hwurzburg, 2026-08-28:
Cygwin is deprecated...closing this

closed 2026-08-28 by Hwurzburg