ArduPilot/ardupilot_wiki #4996: Setting the Aircraft Up for Tuning more specifically "Mission Planner Helper" section

4O6FPV, 2023-03-06:
I think the "Mission Planner Helper" section is not in the right place. On the forum this is often referred to as first time setup and is also titled "initial setup parameters" in mission planner. so why is this hidden away in the tuning section. if this truly is the initial parameter helper then this either need to be put in the initial setup or make it red and its own section like first time setup and initial flight are. I just do not think that the way this  "Mission Planner Helper" is being explained in the forums is in the right section. This needs to standout more and be front and center if we are going to constantly refer people to it in the forums and discord.

Hwurzburg, 2023-03-06:
this helper is NOT an initial setup helper....its specifically for helping setting up initial TUNING parameters, not basic frame and configuration setup,  and belongs in the Tuning Section....where is it being explained in the forums....I will look into the threads and try to see what the confusion is...the Mission Planner title is a bit misleading and really should be Tuning Initial Setup IMO....but its does explicitly state in the box that it is for setting up some battery and prop size influenced params and all other setups in the SETUP tab should be done BEFORE entering data there...The wiki location is appropriate...

4O6FPV, 2023-03-06:
then it needs to not be constantly called an initial setup on the forums and needs to be referred to as initial tuning setup https://discuss.ardupilot.org/t/failing-to-level-during-auto-despite-no-visual-oscillations/97712/5?u=406fpv this is just one example of it being called initial setup and not initial tune.

closed 2023-03-06 by 4O6FPV