ArduPilot/ardupilot_wiki #5009: Complete Parameter List

heinduplessis, 2023-03-11:
It’s mentioned here that the vehicle can only be rudder armed when in manual modes. However we’ve tested many times and one can arm the aircraft in auto mode, causing a very dangerous scenario.

https://ardupilot.org/plane/docs/parameters.html#arming-rudder-arming-with-rudder-enable-disable

Hwurzburg, 2023-03-12:
being discussed at next devcall.... #23186 is being proposed to address...then the param description can be updated in another PR pending acceptance of that PR....even if rejected, I will do a PR to correct the param description

closed 2023-03-12 by heinduplessis

Hwurzburg, 2023-03-14:
results of devcall....my PR added a safety feature to rudder arming in AUTO or MODE TAKEOFF was discussed and some modifications were proposed which I will work on....in the meantime, I will cubmit a PR to correct the param description...