ArduPilot/ardupilot_wiki #5058: Download and Installation of MAVProxy on Raspberry Pi

ravagea1, 2023-04-10:
[Download and Installation](https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html) - Getting this error when trying to download mavproxy to a Raspberry Pi 2 Model B

ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    mavproxy from https://www.piwheels.org/simple/mavproxy/MAVProxy-1.8.59-py3-none-any.whl#sha256=8d96818807009a1ba95fde60746d65c97ddbb495adc527680cb947bd35e04a65:
        Expected sha256 8d96818807009a1ba95fde60746d65c97ddbb495adc527680cb947bd35e04a65
             Got        cfda65018129f8a9391486d3ba3e80fa2ded7b53aed747cf134f843cd39e9899

I copy pasted the code (attached below) into the command line and hit enter and let the installation happen.
```
sudo apt-get install python3-dev python3-opencv python3-wxgtk4.0 python3-pip python3-matplotlib python3-lxml python3-pygame
pip3 install PyYAML mavproxy --user
echo 'export PATH="$PATH:$HOME/.local/bin"' >> ~/.bashrc
```

Any input on how I can fix this? Thanks in advance!

Hwurzburg, 2023-04-10:
This really should be in the support group discuss.ardupilot.org.....you will get the appropriate people looking at it.....I will wait a day or two to close this while you have a chance to move it

ravagea1, 2023-04-11:
Oops sorry about that! Will move it asap

closed 2023-04-11 by ravagea1