ArduPilot/ardupilot_wiki #5063: Provide ROS documentation for ROS 2

Ryanf55, 2023-04-15:
The docs generically say ROS, but now that there are two versions, there should be updated docs. 

Perhaps split them up into ROS 1 and ROS 2 folders?

https://ardupilot.org/dev/docs/ros.html

Then, write the basic docs for using ROS 2, such as how to connect over DDS.

Ryanf55, 2023-06-17:
Yes, we have someone in the discord that volunteered earlier this week. The ROS 2 docs will include information on DDS support, which I can only say, it will be done by the time the next release is, so 4.5. If you want to get started, I created a branch here: https://github.com/Ryanf55/ardupilot_wiki/tree/add-ros2-pi-tutorial

esmanish, 2023-06-17:
Is there anything that I can start working on regarding ROS2 documentation?

Ryanf55, 2023-06-17:
If you know mavlink at all, perhaps you can port this guide to ROS 2? https://ardupilot.org/dev/docs/ros-install.html#installing-mavros

ivanrulik, 2023-06-18:
That is nice. Currently I'm working on the ROS 2 raspberry pi tutorial branch with @Ryanf55. I'll be posting updates over here when I get more progress

closed 2023-11-30 by Hwurzburg