ArduPilot/ardupilot_wiki #5104: Update to ESC CANBus setup for DroneCAN and Piccolo CAN (Currawong)

JacksonUAS, 2023-05-05:
ArduPilot documentation is missing some key information for getting CAN ESC's to work - 

https://ardupilot.org/plane/docs/common-uavcan-setup-advanced.html
https://ardupilot.org/plane/docs/common-velocity-can-escs.html

Piccolo/Currawong -

CAN Node needs to align to motor number, servo channel set in ardupilot is irrelevant
ESC Telemetry is provided under esc number (motor number) .

example -
Node 1 = Motor 1 = ESC Telem 1 = Servo x

There should also be a note that "Listen for Broadcast Messages" needs to be enabled in CEquip for the ESC's to respond to FC commands. 

Note - Currawong ESC's now support DroneCAN as well. 

DroneCAN -

CAN Node ID is irrelevant. ESC index message in DroneCAN GUI needs to align to servo channel associated to the motor number. ESC telemetry is provided as ESC whatever is set to servo channel

example -
Motor 1 = Servo x = ESC Index x = ESC Telem x

So if servo 9 is motor 1, x is 9.

Hwurzburg, 2023-05-08:
@tridge can you confirm and then lets discuss wiki improvements....

SchrodingersGat, 2023-05-16:
@JacksonUAS @tridge we will submit an update to the documentation to address this :)

closed 2023-06-09 by Hwurzburg