ArduPilot/ardupilot_wiki #5133: Error regarding BDSHOT on Aux1

joe-stroup, 2023-05-19:
There is an error in the wiki regarding the use of AUX1 for bi-directional DSHOT.

I've had passthrough work on AUX-1 OK on some ESC's, but fail on others.  However, this does not effect using AUX for BDSHOT - you just have to move the ESC to another AUX port to run pass through for BLHeliSuite32 to configure the ESC.

Also - BDSHOT requires AUX ports to be used in certain groupings.  So the wording of the wiki gives the impression that you could use AUX 2-5 - and as I under stand it, this grouping won't work.

A good resource to verify these details is Andy Piper.

[image]

[image]

[image]

Hwurzburg, 2023-05-23:
@andyp1per  pls confirm and suggest required changes

andyp1per, 2023-05-23:
Yeah I think this is correct, the general problem with AUX1 is that serial passthrough does not work. BDSHOT, however, should work. So just replace "bi-directional dshot" with "serial passthrough"

Hwurzburg, 2023-05-23:
The third paragraph after this links directly to section on grouping..so it's mentioned directly on this page...