ArduPilot/ardupilot_wiki #5146: adding NAMED_VALUE_FLOAT in MissionPlanner

tridge, 2023-05-23:
with more people using scripting to send NAMED_VALUE_FLOAT and the 2nd airspeed currently as "AS2" we should have some docs on how to display those on MissionPlanner
I recently captured these to help a user:
[image]
[image]
[image]
[image]

Hwurzburg, 2023-05-24:
how is named float displayed?

Hwurzburg, 2023-07-01:
Added to MP wiki dealing with USER item display

closed 2023-07-01 by Hwurzburg