ArduPilot/ardupilot_wiki #5148: Adding a New Flight Mode to Copter

HEIseYOUmolc, 2023-05-24:
[Adding a New Flight Mode to Copter](https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I followed the https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html?highlight=adding%20mode method and it does not display properly in QCS

HEIseYOUmolc, 2023-05-24:
I use command sim_vehicle.py -v ArduCopter -f quad --console --map -D 
and then open the QGroundControl and MissionPlanner on ubuntu 20.04LTS

HEIseYOUmolc, 2023-05-24:
By the way ，I want to know the method of modify ParameterFactMetaData.xml is still useful to do this？
I use this method，had nothing to change。My mode always not occur

Hwurzburg, 2023-05-24:
this is really a support issue related to how the GCS mode.xml file is created for each GCS....@peterbarker may be able to help....normally I would close this and redirect to discuss.ardupilot.com...but it aappears that some notes in the DEV wiki about creating the info gcs need to plain text show the mode name might be useful...@peterbarker can you provide some bullets on this?

HEIseYOUmolc, 2023-05-24:
First of all, I am new to Ardupilot, sorry I can't provide you with a lot of information, I will try to post screenshots to make you better solve this problem

HEIseYOUmolc, 2023-05-24:
This is the ardupilotmega.h file compiled by my mavlink build, which mentions that my mode is normally enumerated, and in the previously mentioned guide wiki, GCS may use it to automatically populate the list of available copter modes.

HEIseYOUmolc, 2023-05-25:
> this is really a support issue related to how the GCS mode.xml file is created for each GCS....@peterbarker may be able to help....normally I would close this and redirect to discuss.ardupilot.com...but it aappears that some notes in the DEV wiki about creating the info gcs need to plain text show the mode name might be useful...@peterbarker can you provide some bullets on this?
> 
> you could also do a git-blame on a recent mode like zig-zag or turtle, examine the PR that added it to find the file changes for the GCS emitter

Thanks for your suggestion, I carefully looked at a lot of changes about zigzag_mode on github, although many did not understand, but I think these are changes to refer to .cpp and .h files, similar to the steps on the wiki, I think it should be related to the changes in the xml file on GCS, I have tried some other methods, but still can't show it in mavproxy or QGC or MissionPlanner, The above is the GCS I have used。

[image]

[image]

Hwurzburg, 2023-07-01:
@rmackay9 @peterbarker can you recommend any changes to this document?https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode.html?