ArduPilot/ardupilot_wiki #5161: Tuning for Autonomous Aerobatics --- Document error and a question

normkawakami, 2023-05-30:
1) The "Veryfing the Rate Capabilities" says,
"The [ACRO_PITCH_RATE](https://ardupilot.org/plane/docs/parameters.html#acro-pitch-rate), [ACRO_ROLL_RATE](https://ardupilot.org/plane/docs/parameters.html#acro-roll-rate), and [ACRO_YAW_RATE](https://ardupilot.org/plane/docs/parameters.html#acro-yaw-rate) must be set to less than what the airframe is capable of doing. Values of 360, 180, and 90 degress per second, respectively,"

I believe "360, 180, and 90 degrees" is incorrect and it should be "180, 360 and 90 degrees". 

2) The "Confirming Test Flight" says,
"After verifying and modifying the PIDS, fly again, being ready to resume MANUAL control if unstable. "

My question is in what flight mode should I fly again?

Hwurzburg, 2023-05-30:
the numbers are rates...degrees per second...and the wording is correct

after tuning fly in any stablized mode to check tune

normkawakami, 2023-05-30:
Hwurzburg, Thank you for your response.
I know 360,180 and 90 are rates and expressed in degrees per second.
I still think the pitch rate should be 180 and the roll rates 360, as usually roll is more responsible than pitcth as the graph in the same page shows.

Hwurzburg, 2023-05-30:
True....will  change

Hwurzburg, 2023-07-01:
fixed

closed 2023-07-01 by Hwurzburg