ArduPilot/ardupilot_wiki #5162: Copter Commands in Guided Mode

DonatelloX, 2023-05-31:
In the SET_POSITION_TARGET_LOCAL_NED command, there is an error.
I read `type_mask `in the _Command Field_:
`bit1:PosX, bit2:PosY, bit3:PosZ, bit4:VelX, bit5:VelY, bit6:VelZ, bit7:AccX, bit8:AccY, bit9:AccZ, bit11:yaw, bit12:yaw rate`
but in example is written:
`Use Position : 0b110111111000 / 0x0DF8 / 3576 (decimal)`
It should be:
`0b111000000000`

Hwurzburg, 2023-07-12:
@rmackay9 pls review

Hwurzburg, 2023-08-31:
@rmackay9 bump

Hwurzburg, 2023-11-30:
@rmackay9 bump again

rmackay9, 2023-12-14:
Oops sorry, I'll handle this.

rmackay9, 2023-12-14:
Hi @DonatelloX,

The typemask field is super annoying and leads to confusion because each bit is actually an ignore bit ([see here](https://mavlink.io/en/messages/common.html#POSITION_TARGET_TYPEMASK)).

So I think what we have in the wiki is correct and you can always prove this by copy-pasting the example into MAVProxy's console.

Thanks again for the report and apologies to all for the slow response.

closed 2023-12-14 by rmackay9