ArduPilot/ardupilot_wiki #5164: Valgrind dependency not mentioned in setup instructions

slim71, 2023-06-02:
I've cloned the repo and set the build environment following every step in the [related guide](https://ardupilot.org/dev/docs/building-setup-linux.html).

I then had the need to run a **Valgrind** analysis (during a debug session), for which I used the command
```bash
sim_vehicle.py -v ArduPlane -DV --console --enable-dds
```
but it resulted in the errors below.
The problem is that I learnt of the need for Valgrind when actually using the code, at which point I didn't have it installed because it is nowhere mentioned as _at least_ a possible requirement. 

Adding it might be a good idea.

XTerm screenshot
[image]

sim_vehicle.py output
```bash
slim71@slim71-Ubuntu:~/Documents/git/ardupilot$ sim_vehicle.py -v ArduPlane -DV --console --enable-dds
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/slim71/Documents/git/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl" "--debug" "--enable-dds"
Setting top to                           : /home/slim71/Documents/git/ardupilot 
Setting out to                           : /home/slim71/Documents/git/ardupilot/build 
Autoconfiguration                        : enabled 
Checking for program 'python'            : /usr/bin/python 
Checking for python version >= 3.6.9     : 3.10.6 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 11.3.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Checking for program 'microxrceddsgen'   : /home/slim71/Documents/git/Micro-XRCE-DDS-Gen/scripts/microxrceddsgen 
Enabled OpenDroneID                      : no 
Enabled firmware ID checking             : no 
GPS Debug Logging                        : no 
Enabled custom controller                : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
DC_DSDL compiler                               : /home/slim71/Documents/git/ardupilot/modules/DroneCAN/dronecan_dsdlc/dronecan_dsdlc.py 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Debug build                                    : enabled 
Coverage build                                 : disabled 
Force 32-bit build                             : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.433s)
{'waf_target': 'bin/arduplane', 'default_params_filename': 'models/plane.parm', 'model': 'plane', 'sitl-port': True}
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/slim71/Documents/git/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arduplane"
Waf: Entering directory `/home/slim71/Documents/git/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Embedding file models/plane-3d.parm:Tools/autotest/models/plane-3d.parm
Embedding file models/plane.parm:Tools/autotest/models/plane.parm
Embedding file models/xplane_heli.json:Tools/autotest/models/xplane_heli.json
Embedding file models/xplane_plane.json:Tools/autotest/models/xplane_plane.json
building /home/slim71/Documents/git/ardupilot/build/sitl/modules/Micro-XRCE-DDS-Client/include/uxr/client/config.h
building /home/slim71/Documents/git/ardupilot/build/sitl/modules/Micro-CDR/include/ucdr/config.h
Waf: Leaving directory `/home/slim71/Documents/git/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/slim71/Documents/git/ardupilot/build/sitl
Target         Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)  External Flash Used (B)
---------------------------------------------------------------------------------------------------------
bin/arduplane   4857285    177016   209896               5034301  Not Applicable  Not Applicable         

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (2.336s)
SIM_VEHICLE: Using defaults from (Tools/autotest/models/plane.parm)
SIM_VEHICLE: Run ArduPlane (valgrind)
SIM_VEHICLE: "/home/slim71/Documents/git/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane (valgrind)" "valgrind" "--soname-synonyms=somalloc=nouserintercepts" "--track-origins=yes" "/home/slim71/Documents/git/ardupilot/build/sitl/bin/arduplane" "-S" "--model" "plane" "--speedup" "1" "--slave" "0" "--defaults" "Tools/autotest/models/plane.parm" "--sim-address=127.0.0.1" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--console"
RiTW: Starting ArduPlane (valgrind) : valgrind --soname-synonyms=somalloc=nouserintercepts --track-origins=yes /home/slim71/Documents/git/ardupilot/build/sitl/bin/arduplane -S --model plane --speedup 1 --slave 0 --defaults Tools/autotest/models/plane.parm --sim-address=127.0.0.1 -I0
Connect tcp:127.0.0.1:5760 source_system=255
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
Failed to connect to tcp:127.0.0.1:5760 : [Errno 111] Connection refused
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks
```

Hwurzburg, 2023-07-09:
@peterbarker bump for review

peterbarker, 2023-07-27:
> sim_vehicle.py -v ArduPlane -DV --console --enable-dds

Would https://github.com/ArduPilot/ardupilot/pull/24416 have prevented your problems?  i.e. did you use the standard environment setup scripts to create your environment?

slim71, 2023-07-28:
It's been a while, but if I remember correctly I have, since I followed [this guide](https://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux).
So adding it there would have prevented the problem, yes.

Ryanf55, 2023-07-28:
> > sim_vehicle.py -v ArduPlane -DV --console --enable-dds
> 
> Would [ArduPilot/ardupilot#24416](https://github.com/ArduPilot/ardupilot/pull/24416) have prevented your problems? i.e. did you use the standard environment setup scripts to create your environment?

Yes, approved!

peterbarker, 2023-07-29:
Thanks @Ryanf55 , @slim71 , I've merged that PR.

closed 2023-07-29 by peterbarker