ArduPilot/ardupilot_wiki #5272: Setting up SITL using Vagrant does not work for default machine

SaschaZell, 2023-07-11:
I'm following the instructions in [Setting up SITL using Vagrant](https://ardupilot.org/dev/docs/setting-up-sitl-using-vagrant.html) on my Mac and after running command

`vagrant ssh -c "sim_vehicle.py -j 2 -v ArduCopter"`

I get the error message:

[image]

Using another machine name (e.g. jammy) `vagrant up jammy` and `vagrant ssh jammy -c "sim_vehicle.py -j 2 -v ArduCopter"` seems to work fine. I'm very new to vagrant and I can only speculate that it has something to do with the python interpreter defined in file _install-prereqs-ubuntu.sh_

peterbarker, 2023-07-11:
Something is amiss here.  @SaschaZell presumably followed the instructions to bring up a Vagrant VM and it doesn't actually build ArduPilot.  PR incoming.

peterbarker, 2023-07-11:
@SaschaZell please retry - you ought to be able to do a simple `git pull` in your checkout to get the fixes, but if that doesn't get you going then blowing everything away and starting fresh might be the go :-)