ArduPilot/ardupilot_wiki #5304: Sensor Position Offset Compensation documentation is confusing

paulr4, 2023-07-27:
The [Sensor Position Offset Compensation](https://ardupilot.org/copter/docs/common-sensor-offset-compensation.html#common-sensor-offset-compensation) page gives some useful detail, but uses IMU position and centre-of-gravity interchangeably.

For example:

"[GPS_POS1_X](https://ardupilot.org/copter/docs/parameters.html#gps-pos1-x), [GPS_POS1_Y](https://ardupilot.org/copter/docs/parameters.html#gps-pos1-y), [GPS_POS1_Z](https://ardupilot.org/copter/docs/parameters.html#gps-pos1-z) the first GPS’s position from the vehicle’s IMU or center-of-gravity"

What if the C of G and IMU are not co-located? Surely one or the other is the correct reference point, but the documentation does not make it clear.

rmackay9, 2023-07-27:
Yes indeed, it's a little unclear.  I think that the X, Y, Z offsets are relative to the IMU if the user hasn't set the INS_POS_XYZ parameters... but relative to the vehicle's COG if INS_POS_XYZ have been set.  This is what @priseborough told me (I think).

Hwurzburg, 2023-07-27:
when @priseborough confirms as correct, @rmackay9 , I will update the wiki page

priseborough, 2023-07-30:
That is correct. If the IMU position is left at 0,0,0 then by default it is at the body frame origin and everything else can be measured wrt the IMU

Hwurzburg, 2023-07-31:
added note

closed 2023-07-31 by Hwurzburg