ArduPilot/ardupilot_wiki #5388: Add ESP32 autopilot setup page

rmackay9, 2023-09-04:
We have this very sparse page https://ardupilot.org/dev/docs/esp32-autopilot.html but  we have a [request on the forums](https://discuss.ardupilot.org/t/copter-4-4-0-released/105233/46) here for more complete documentation.

Hwurzburg, 2023-09-04:
@rmackay9 who to assign this to?

rmackay9, 2023-09-04:
@Hwurzburg I think it should be @davidbuzz but I'm not completely sure.

davidbuzz, 2023-10-11:
esp32 "port" is experimental, and has known issues. I do not support user-facing documentation at this time, the existing above 'warning' is correct, and the request in the forum should be pointed to the 'esp32' discord channel where the *ongoing development* happens.   major issues affecting esp32: main loop time/s are not fast enough, parameter download over wifi is super slow, there are no ready-to-buy boards being made by any manufacturer, so every-one has to wire their own,  doesn't bootup as fast as it should, in some circumstances mavftp doesn't work,  very small number of imu's are tested [ basically just a gy-91] , very limited io available so careful selection of gpio's used is needed, and more.

closed 2023-10-11 by davidbuzz