ArduPilot/ardupilot_wiki #5432: Mission Planning

larrypatterson, 2023-09-17:
[Mission Planning](https://ardupilot.org/planner/docs/common-mission-planning.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

larrypatterson, 2023-09-17:
I really need help!  The movement of servos does go thru the pixhawk controller.  Movement is good on the computer mission planner screen.  But no Gyro movements connected to Pixhawk.  I would really appreciate your help   Thank you  Larry

larrypatterson, 2023-09-17:
I really need help! The movement of servos does go thru the pixhawk controller. Movement is good on the computer mission planner screen. But no Gyro movements connected to Pixhawk. I would really appreciate your help Thank you Larry
My email is cedarvalleyairport@gmail.com

Hwurzburg, 2023-09-17:
try discuss.ardupilot.com for support issues...and read the two red highlighted sections of the wiki for your vehicle

larrypatterson, 2023-09-18:
Hi Henry
This is my first experience with pixhawk.  I bought two of the 145 dollar
2.4.8 units.    I have spent several weeks trying to get one of the pixhawk
to give me an out put to servos.
I have tried putting 5 volts servo panel.
I read thru the BRD safety mast several times along with the arduupilot.com.
  AT 75 years old i am just not very sharp anymore. I have gotten pretty
good about downloading stuff.
there use to be an airplane which flew around on the simulator.  Lost that
plane.  Do you know how I could get the simulator plane back?   I have
tried deleting everything several times, and reloading.
several times.  Didnt help.
  I do appreciate your help.!

Hwurzburg, 2023-11-30:
if still having issue, raise in support forum...

closed 2023-11-30 by Hwurzburg