ArduPilot/ardupilot_wiki #5452: ROS 2

ShubhajyotiBarman, 2023-09-28:
unix_explore@Unidentified-Network:~/ros2_ws$ cd ~/ros2_ws
colcon build --cmake-args -DBUILD_TESTING=ON
Starting >>> micro_ros_agent
Starting >>> ardupilot_msgs
Starting >>> microxrcedds_agent
Starting >>> mrc
Finished <<< mrc [0.58s]
Finished <<< micro_ros_agent [0.69s]
Starting >>> ardupilot_sitl
Finished <<< microxrcedds_agent [0.94s]
Finished <<< ardupilot_msgs [1.53s]
--- stderr: ardupilot_sitl
invalid lock file in /home/unix_explore/ros2_ws/src/ardupilot
Could not find the program ['microxrceddsgen']
(complete log in /home/unix_explore/ros2_ws/src/ardupilot/build/config.log)
gmake[2]: *** [CMakeFiles/ardupilot_configure.dir/build.make:70: CMakeFiles/ardupilot_configure] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/ardupilot_configure.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ardupilot_sitl [2.44s, exited with code 2]

Summary: 4 packages finished [3.78s]
  1 package failed: ardupilot_sitl
  1 package had stderr output: ardupilot_sitl
  1 package not processed

pedro-fuoco, 2023-10-24:
> Could not find the program ['microxrceddsgen']

Did you follow the "Build microxrcedds_gen" section in the installation page?
https://ardupilot.org/dev/docs/ros2.html#ros2

Hwurzburg, 2023-11-30:
no response to @pedro-fuoco  question... assuming resolved...closing issue

aarsht7, 2024-04-18:
> > Could not find the program ['microxrceddsgen']
> 
> Did you follow the "Build microxrcedds_gen" section in the installation page? https://ardupilot.org/dev/docs/ros2.html#ros2

Hello @pedro-fuoco ,

I encountered the same error. Yes, I did follow the installation for the `microxrcedds_gen` 
here is the output for the version info

```bash
ros@aarsh-pc:/workspaces/ardupilot$  microxrceddsgen -version
openjdk version "11.0.22" 2024-01-16
OpenJDK Runtime Environment (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1)
OpenJDK 64-Bit Server VM (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1, mixed mode, sharing)
microxrceddsgen version: 2.0.2
```

Error while building the `ardupilot_sitl`
```bash
Starting >>> ardupilot_sitl
--- stderr: ardupilot_sitl                         
gmake[2]: *** [CMakeFiles/ardupilot_build.dir/build.make:70: CMakeFiles/ardupilot_build] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:168: CMakeFiles/ardupilot_build.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ardupilot_sitl [1.57s, exited with code 2]
```

Is it a known issue?
I tried to look up to the internet but no lead. Do you have any fixes?