ArduPilot/ardupilot_wiki #5454: Fences

wagatan, 2023-09-29:
[Fences](https://ardupilot.org/copter/docs/common-geofencing-landing-page.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Bit type is updated .
I checked copter V.4.3.7

1	Max altitude
2	Circle Centered on Home
4	Inclusion/Exclusion Circles + Polygons
8	Min altitude

Thanks

Hwurzburg, 2023-09-29:
the wiki appears correct....what is the issue? titles are clear and correct although do not exactly match abbreviated labels in parameter metadata

wagatan, 2023-09-29:
[bit.pdf](https://github.com/ArduPilot/ardupilot_wiki/files/12763950/bit.pdf)
I attached a pdf file.
Please check it.

Thanks

Hwurzburg, 2023-09-29:
[image]
and here is the wiki table...which conveys the same information
what is confusing you is that it is a BITMASK...sitting BIT 2 adds "4" to the VALUE

closed 2023-09-29 by Hwurzburg