ArduPilot/ardupilot_wiki #5463: RPLidar A2 360 degree lidar "Prearm: PRX1: No Data"

turkanispak, 2023-10-04:
[RPLidar A2 360 degree lidar](https://ardupilot.org/copter/docs/common-rplidar-a2.html) - 
Hello,
I have recently bought RPLidar A2 360 degree lidar (A2M12 specifically) to implement obsracle detection avoidance. I have verified that lidar works using RoboStudio_v1.8.1 app. I have followed the instructions on [https://ardupilot.org/copter/docs/common-rplidar-a2.html](url) and added a seperate BEC to feed the lidar. 

**Lidar starts to spin when power is on but I am unable to read data from proximity window and I get the "Prearm: PRX1: No Data" error.**

Other than feeding power from an external source I have tried switching powering orders to both pixhawk and rplidar (ie. power lidar first then reboot pixhawk) but I've had no luck.

Any help is greatly appreciated.

turkanispak, 2023-10-24:
Hello,

I have solved the issue by setting setting **"SERIAL1_BAUD=256"**, it took me couple of hours because I somehow thought that RPLidar A2M12 would automatically work with 115000 baud rate. I could not figure out how to change lidar's baud rate so I changed the SERIAL1_BAUD parameter.

Thank you all for your time.

Hwurzburg, 2023-10-24:
@rishabsingh3003 are any wiki changes needed?

rishabsingh3003, 2023-10-26:
@Hwurzburg as far as I can looking at the datasheet that the default baud rate is 115200. So this documentation should be good. Thanks

closed 2023-10-26 by rishabsingh3003