ArduPilot/ardupilot_wiki #5566: Docker installation defaults to sudo.

wtokumaru, 2023-11-16:
The docker installation steps here put you in a container that has sudo: https://ardupilot.org/dev/docs/ros2.html#installation-docker
This contradicts the SITL setup requiring not to be run as sudo if you do it in the container: https://ardupilot.org/dev/docs/building-setup-linux.html#install-some-required-packages

Am still able to install through other workflows, so this is not blocking, but might not be intended.

Ryanf55, 2023-11-16:
This will affect non-ROS2 users also. The default developer dockerfile has the same problem because ArduPilot has users run it as root.

Ryanf55, 2023-11-16:
@khancyr I'd love your input. Do we want ArduPilot devs using docker as root, and make ardupilot scripts capable of running in root, or do we want to add stuff to all our dockerfiles to make them not root?

khancyr, 2023-11-16:
Those docker image where intend for CI not generic usage that is why they are using root user and have minimal stuff.

We should build a better image like the one on ArduPilot repo that have proper permission and generic install step