ArduPilot/ardupilot_wiki #5588: 'Tuning Turn Rate' page should mention MOT_SPD_SCA_BASE

calebstott, 2023-11-24:
[Tuning Turn Rate](https://ardupilot.org/rover/docs/rover-tuning-steering-rate.html) 

During the tuning of a ground rover (running 4.5.0-dev) we spent a lot of time trying to figure out why we couldn't get our rover to steer properly while in auto modes. Was constantly overshooting waypoints

Turned out [MOT_SPD_SCA_BASE](https://ardupilot.org/rover/docs/parameters.html#mot-spd-sca-base-motor-speed-scaling-base-speed) was set at default of 1 m/s which was too slow for our setup

Would be handy if there was a mention on the tuning guide that it may need adjusting for some vehicles

rmackay9, 2023-11-24:
Hi @calebstott,

Thanks for the report.  We can certainly add a note on this.  I'm a little surprised that raising this was necessary though.  If you post a log in the forums I'm happy to have a look.

rmackay9, 2024-01-14:
We haven't heard back on this so I think I will close this.

closed 2024-01-14 by rmackay9