ArduPilot/ardupilot_wiki #5608: Searchfatal: not a git repository: ../../.git/modules/modules/PX4Firmware fatal: could not reset submodule index

skjerr, 2023-11-30:
[Search](https://ardupilot.org/dev/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Hwurzburg, 2023-11-30:
will need a lot more info on what the issue is...this is not an ArduPilot module

tbrookin, 2024-01-02:
+1 on this.  Seems like the repo is in a bad state.  Repro steps:

1. Clone the ardupilot repo.  (local will be set to master branch)
2. Try to checkout a subbranch on the local (I used origin\Plane-4.4)

Checkout fails with this git error:

fatal: not a git repository: ../../.git/modules/modules/uavcan
fatal: could not reset submodule index

@Hwurzburg

Hwurzburg, 2024-01-02:
did you clone using the command given in the wiki?

git clone --recurse-submodules https://github.com/your-github-userid/ardupilot

also cloning does not fetch any branch other than master....to get other branchs down locally to checkout you need to git fetch origin...a git branch -r what remote branches are local

rmackay9, 2024-01-03:
I don't think the PX4Firmware module is used anymore.