ArduPilot/ardupilot_wiki #5685: powerflag 0x16 MAV_POWER_STATUS_PERIPH_HIPOWER_OVERCURRENT on Pixhawk 6c

prima-bez, 2023-12-28:
ArduCopter V4.3.2 (fcef79e7) 
ChibiOS: 66e5de0d
Pixhawk 6c connected to PC via USB

ER
voltageflag = 4 MAV_POWER_STATUS_USB_CONNECTED=4, / USB power is connected | 

AR
voltageflag = 20 

(MAV_POWER_STATUS_USB_CONNECTED=4, / USB power is connected | 
MAV_POWER_STATUS_PERIPH_HIPOWER_OVERCURRENT=16, / hi-power peripheral supply is in over-current state | /

AI_1
# power sensing
PE3 VDD_5V_PERIPH_nOC INPUT PULLUP
PF13 VDD_5V_HIPOWER_nOC INPUT PULLUP

But STM32H743 hasn't a pin PF13
FAULT IC bq24315DSG connected to B8 (PC11)

PE2 (A3) - nVDD_5V_PERIPH_EN  OUTPUT HIGH
PE3 (B3) - VDD_5V_PERIPH_nOC INPUT PULLUP
PC10 (B9) - nVDD_5V_HIPOWER_EN OUTPUT HIGH
PC11 (B8) - VDD_5V_HIPOWER_nOC INPUT PULLUP

[image]
[image]

Hwurzburg, 2023-12-28:
this is not a wiki issue...please raise this either in the discuss.ardupilot.org support forum or in the HELP chats on ardupilot discord...

fyi BGA stm32h743's do have PF13 pin...only the 100 pin version does not have it

closed 2023-12-28 by Hwurzburg

prima-bez, 2023-12-28:
Hello! thank you)

what about stm, pixhawk contain exactly 100 pin  bga version)

Hwurzburg, 2023-12-28:
you are correct...raise an issue here for the Pixhawk6C hwdef : https://github.com/ArduPilot/ardupilot/issues