ArduPilot/ardupilot_wiki #5696: Add known working RPM sensors to Heli RPM sensor wiki

rmackay9, 2024-01-07:
We should add this RPM sensor to the RPM sensor wiki page as a known working model https://www.controlhobbies.com/Telemetry-magnetic-sensor-to-measure-RPM-33616.html

https://ardupilot.org/copter/docs/common-rpm.html

Hwurzburg, 2024-01-07:
are there any others @bnsgeyer @Gone4Dirt ?

rmackay9, 2024-01-11:
This seems to be the only other one that we know about.  I asked on the trad-heli discord channel..
https://www.sm-modellbau.de/magnetischer-Drehzahlsensor-fuer-UniTest-2-UniLog-1-2

Gone4Dirt, 2024-01-11:
Sorry for being slow.  
I know that this works:
https://modelhelicopters.co.uk/products/align-beastx-governor-sensor-hegbp002t?variant=42819003220168
Requires a pull up resistor (between 5V and signal).

Hwurzburg, 2024-01-14:
done

closed 2024-01-14 by Hwurzburg