ArduPilot/ardupilot_wiki #5701: Using SITL with AirSim -- AirSim ended

patrickpoirier51, 2024-01-09:
[Using SITL with AirSim](https://ardupilot.org/dev/docs/sitl-with-airsim.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

https://onmsft.com/news/microsoft-abruptly-axes-metaverse-project-triggers-staff-layoffs/

Project Airsim, an AI-based aircraft simulation system aimed at industrial applications, was expected to be a key component of Microsoft’s vision for the “industrial metaverse.” The system, designed to teach individuals how to react to various flight scenarios, has been shut down. The move announced through a company statement to[ Business Insider](https://www.businessinsider.com/microsoft-shutters-project-airsim-ai-strategy-openai-2023-10?IR=T), has resulted in the layoff of the entire team associated with the project,

khancyr, 2024-01-09:
There are 2 things : Airsim that is probably abandonned for some time already but open source and that we support. And the new version Project Airsim that was subscription only and not compatible with ArduPilot and not supported by us.

patrickpoirier51, 2024-01-09:
I think we should add a warning on wiki about the End of Life of product as a result  it will be  harder to support and configure with UnReal or Unity Engines that are less and less compatible with the open source release as time goes by.

Hwurzburg, 2024-01-09:
need proposed wording of note