ArduPilot/ardupilot_wiki #5743: Build wiki failed, some kind of connection issue?

lida2003, 2024-01-29:
Just follow [Setup the Environment in Ubuntu 22.04 LTS](https://ardupilot.org/dev/docs/common-wiki-editing-setup.html)

It's OK with `Sphinxsetup.sh` script.
```
$ ./Sphinxsetup.sh
```

Build wiki failed, googled around, I didn't figure out what's wrong. Any idea?

```
$ python3 update.py --site copter
/home/daniel/SnapDragonfly/ardupilot_wiki/update.py:769: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Fetching https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/scripts/build_options.py
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 169, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 96, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 86, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 700, in urlopen
    httplib_response = self._make_request(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 383, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 1017, in _validate_conn
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 353, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7f99e146f190>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 756, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /ArduPilot/ardupilot/master/Tools/scripts/build_options.py (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f99e146f190>: Failed to establish a new connection: [Errno 111] Connection refused'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 1033, in <module>
    create_features_pages(args.site)
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 811, in create_features_pages
    fetch_url("https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/scripts/build_options.py")
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 153, in fetch_url
    total_size = get_request_file_size(fetchurl)
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 183, in get_request_file_size
    hresponse = requests.head(url, headers=headers)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/api.py", line 100, in head
    return request("head", url, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /ArduPilot/ardupilot/master/Tools/scripts/build_options.py (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7f99e146f190>: Failed to establish a new connection: [Errno 111] Connection refused'))

```

Hwurzburg, 2024-01-29:
it appears that your network is not allowing a secure connection to the ArduPilot repos....try again later or you need to contact your internet provider

lida2003, 2024-01-30:
> it appears that your network is not allowing a secure connection to the ArduPilot repos....try again later or you need to contact your internet provider

wget through https is OK, but it seems not ok with https git url.

```
Successfully installed Jinja2-3.1.3 Pygments-2.17.2 alabaster-0.7.16 babel-2.14.0 charset-normalizer-3.3.2 docutils-0.18.1 imagesize-1.4.1 packaging-23.2 requests-2.31.0 snowballstemmer-2.2.0 sphinx-7.2.6 sphinxcontrib-applehelp-1.0.8 sphinxcontrib-devhelp-1.0.6 sphinxcontrib-htmlhelp-2.0.5 sphinxcontrib-jsmath-1.0.1 sphinxcontrib-qthelp-1.0.7 sphinxcontrib-serializinghtml-1.1.10
+ python3 -m pip install --user --upgrade lxml
Collecting lxml
  Downloading lxml-5.1.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.5 kB)
Downloading lxml-5.1.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (8.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.0/8.0 MB 401.0 kB/s eta 0:00:00
Installing collected packages: lxml
Successfully installed lxml-5.1.0
+ python3 -m pip install --user --upgrade git+https://github.com/ArduPilot/sphinx_rtd_theme.git
Collecting git+https://github.com/ArduPilot/sphinx_rtd_theme.git
  Cloning https://github.com/ArduPilot/sphinx_rtd_theme.git to /tmp/pip-req-build-_55ijer3
  Running command git clone --filter=blob:none --quiet https://github.com/ArduPilot/sphinx_rtd_theme.git /tmp/pip-req-build-_55ijer3
  fatal: unable to access 'https://github.com/ArduPilot/sphinx_rtd_theme.git/': Failed to connect to github.com port 443 after 135244 ms: Connection timed out
  error: subprocess-exited-with-error

  × git clone --filter=blob:none --quiet https://github.com/ArduPilot/sphinx_rtd_theme.git /tmp/pip-req-build-_55ijer3 did not run successfully.
  │ exit code: 128
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet https://github.com/ArduPilot/sphinx_rtd_theme.git /tmp/pip-req-build-_55ijer3 did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

```

lida2003, 2024-01-30:
Well, wget through https is OK, but `python3 update.py --site copter` failed again.

```
daniel@daniel-ThinkPad-SL410:~/SnapDragonfly/ardupilot_wiki$ wget -O opencv.zip https://github.com/opencv/opencv/archive/refs/tags/4.0.0.zip
--2024-01-30 16:05:34--  https://github.com/opencv/opencv/archive/refs/tags/4.0.0.zip
Connecting to 192.168.1.13:808... connected.
Proxy request sent, awaiting response... 302 Found
Location: https://codeload.github.com/opencv/opencv/zip/refs/tags/4.0.0 [following]
--2024-01-30 16:05:35--  https://codeload.github.com/opencv/opencv/zip/refs/tags/4.0.0
Connecting to 192.168.1.13:808... connected.
Proxy request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘opencv.zip’

opencv.zip                                    [   <=>                                                                                ] 713.18K  1.09MB/s               ^C
daniel@daniel-ThinkPad-SL410:~/SnapDragonfly/ardupilot_wiki$ python3 update.py --site copter
/home/daniel/SnapDragonfly/ardupilot_wiki/update.py:769: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Fetching https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/scripts/build_options.py
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 697, in urlopen
    self._prepare_proxy(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 971, in _prepare_proxy
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 411, in connect
    self.sock = ssl_wrap_socket(
  File "/usr/lib/python3/dist-packages/urllib3/util/ssl_.py", line 449, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(
  File "/usr/lib/python3/dist-packages/urllib3/util/ssl_.py", line 493, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.10/ssl.py", line 513, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.10/ssl.py", line 1100, in _create
    self.do_handshake()
  File "/usr/lib/python3.10/ssl.py", line 1371, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 756, in urlopen
    retries = retries.increment(
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /ArduPilot/ardupilot/master/Tools/scripts/build_options.py (Caused by SSLError(SSLEOFError(8, '[SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 1033, in <module>
    create_features_pages(args.site)
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 811, in create_features_pages
    fetch_url("https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/scripts/build_options.py")
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 153, in fetch_url
    total_size = get_request_file_size(fetchurl)
  File "/home/daniel/SnapDragonfly/ardupilot_wiki/update.py", line 183, in get_request_file_size
    hresponse = requests.head(url, headers=headers)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/api.py", line 100, in head
    return request("head", url, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/daniel/.local/lib/python3.10/site-packages/requests/adapters.py", line 517, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /ArduPilot/ardupilot/master/Tools/scripts/build_options.py (Caused by SSLError(SSLEOFError(8, '[SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1007)')))
daniel@daniel-ThinkPad-SL410:~/SnapDragonfly/ardupilot_wiki$
```

lida2003, 2024-01-30:
@Hwurzburg  It works now! It seems git through https might be not OK here, behind  great fire wall.

closed 2024-01-30 by lida2003