ArduPilot/ardupilot_wiki #5759: 4.6 Wiki (should be in Upcoming features)

Hwurzburg, 2024-02-02:


David-OConnor, 2024-02-03:
LMK if there's anything I can do to help, or if this is something you generally handle.

Hwurzburg, 2024-02-03:
I got it...thanks

Ryanf55, 2024-03-24:
26131 will be backported to 4.5 FYI.

LaneaLucy, 2024-04-06:
For 26568 Relay invert there is some explanation on discord

mbuzdalov, 2024-05-16:
26321 is addressed here: https://github.com/ArduPilot/ardupilot_wiki/pull/5994

It covers the EDTv2 contribution, but also explains the basic extended DShot telemetry, because I could not find it anywhere.

Hwurzburg, 2024-05-17:
@mbuzdalov thanks

joshanne, 2024-05-22:
@Hwurzburg 26623 was a source documentation update to reflect what was actually already occurring in code.

The improved text around the text was included in the wiki PR: https://github.com/ArduPilot/ardupilot_wiki/pull/5841

Hwurzburg, 2024-05-22:
> @Hwurzburg 26623 was a source documentation update to reflect what was actually already occurring in code.
> 
> The improved wiki documentation around the bitmask was included in the wiki PR: #5841
great...then done

lida2003, 2024-05-23:
26812 Aocoda-RC H743Dual motor 5-8 mis-labeled issue(@lida2003 ) #6010

rahulOCR, 2024-09-18:
[26886](https://github.com/ArduPilot/ardupilot/pull/26886) MLX96014 Temp Sensor #6286

closed 2024-11-16 by Hwurzburg