ArduPilot/ardupilot_wiki #5779: Joystick/Gamepad

bulentkipcak, 2024-02-10:
Hi, 
I set it up to be able to use the joystick and enabled control with the joystick i can change flight modes but it doesn't react to throttle and direction controls. When i made the same settings in my windows11 i could use it without any problems but i cannot use it in ubuntu 20.04.6. i am currently using the joystick with QGroundControl without any problems but i cannot use it in mission planner.

when i use the joystick the bars in the menu below react
[image]

but there is no change in the bars in this menu
[image]

Hwurzburg, 2024-02-10:
joystick is NOT RC..its MAVLink override messages....joystick movements will not appear on an RC setup screen....and RCx_MIN-MAX params have no impact on joystick operation...it must  be calibrated using that calibration screen

bulentkipcak, 2024-02-10:
> joystick is NOT RC..its MAVLink override messages....joystick movements will not appear on an RC setup screen....and RCx_MIN-MAX params have no impact on joystick operation...it must be calibrated using that calibration screen

i did what it said in the documentation

bulentkipcak, 2024-02-11:
still cannot solved

rmackay9, 2024-02-11:
Hi @Hwurzburg,

FYI, when using MP once a joystick is enabled it will indeed update on the RC calibration screen.

@bulentkipcak,

I've used this quite recently and it does work so I'm going to close this issue.  By the way, it is best to raise issues in [the support forums](https://discuss.ardupilot.org/c/arducopter/copter-44/169) instead of here which is really our to-do list which is reserved for confirmed bugs and enhancement requests.  Sorry to be strict but otherwise our to-do list turns into another support forum.

closed 2024-02-11 by rmackay9

Hwurzburg, 2024-02-12:
@rmackay9 yeah sorry, confused NOT using that cal screen with a joystick to set limits(which you should not do), with it not moving the bars`