ArduPilot/ardupilot_wiki #5781: Sponsors page has 'nan' for one link

ShaneCurcuru, 2024-02-12:
I've been building a database of sponsors for interesting FOSS foundations, and parsing your https://ardupilot.org/ardupilot/docs/common-partners.html page found an <a ... href="nan"> link that obviously doesn't actually go to a sponsor page.

This change was introduced back in 2022 - I'm not sure if it was intentional or not.
https://github.com/ArduPilot/ardupilot_wiki/commit/0e7285323db979bdc4110c887deba0c5c2396210#diff-af070d1dd897e061b67c6122d3a3ca5f2321262ea707fc1e80fa848c4e69b351R556

Feel free to close if this was intentional, I was just curious (and now I need to cut out any 'nan' hrefs from my report).

Hwurzburg, 2024-02-14:
@rishabsingh3003 if they are still a partner, will you fix the target link, or do you want me to?

rishabsingh3003, 2024-02-14:
@Hwurzburg not all Partners have a website, and this one does not (atleast till 6 months back they didn't). The "nan" was intentional from my end.. but really feel free to change it to anything you'd like

closed 2024-02-14 by rishabsingh3003