ArduPilot/ardupilot_wiki #5838: BUG: MAVExplorer setup docs don't work on Ubuntu 22

Ryanf55, 2024-03-24:
Per [this wiki](https://ardupilot.org/dev/docs/using-mavexplorer-for-log-analysis.html#installing-mavexplorer-on-linux), the instructions don't work.

Instructions:
```bash
sudo apt-get install python-matplotlib python-serial python-wxgtk3.0 python-lxml
sudo apt-get install python-scipy python-opencv  python-pip python-pexpect python-tk
sudo pip install --upgrade pymavlink mavproxy
```

Results on Ubuntu 22.04:
```bash
MAV> ryan@B650-970:~/Dev/ros2_ws/src/ardupisudo apt-get install python-matplotlib python-serial python-wxgtk3.0 python-lxml-lxml
[sudo] password for ryan: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-matplotlib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'python-matplotlib' has no installation candidate
E: Unable to locate package python-serial
E: Unable to locate package python-wxgtk3.0
E: Couldn't find any package by glob 'python-wxgtk3.0'
E: Couldn't find any package by regex 'python-wxgtk3.0'
E: Unable to locate package python-lxml

```

khancyr, 2024-03-27:
ardupilot install script is handling this. Probably need to update the wiki to update the packages

Ryanf55, 2024-03-27:
> ardupilot install script is handling this. Probably need to update the wiki to update the packages

Thanks. Do you think we should say for Linux to just run the installer?

khancyr, 2024-04-02:
we should probably add an install script for mavproxy, that would be easier to maintain. On side not, you don't need to install wx, opencv, and matplotlib if you don't need GUI stuffs

tridge, 2024-07-15:
this should be right: ```
sudo apt-get install python3-matplotlib python3-serial python3-wxgtk4.0 python3-lxml
sudo apt-get install python3-scipy python3-opencv  python3-pip python3-pexpect python3-tk
```

Hwurzburg, 2024-07-16:
wiki changed

Ryanf55, 2024-07-16:
Nice, LGTM. 
https://ardupilot.org/dev/docs/using-mavexplorer-for-log-analysis.html#installing-mavexplorer-on-linux

closed 2024-07-16 by Ryanf55

dakejahl, 2025-01-02:
I am having issues using MAVExplorer (first time user). MAVExplorer.py will launch but it fails to load any graphs. I followed the steps in the wiki link above and also tried tridges suggestion above, the only difference being wxgtk4.0 vs 3.0

I am on Ubuntu 22.04. I've tried a some variations of package versions without any luck. Does anyone have a list of package versions from a functional installation? 

**package versions**
```
lxml                                 4.9.3
matplotlib                           3.10.0
matplotlib-inline                    0.1.3
pexpect                              4.8.0
pymavlink                            2.4.40
pyserial                             3.5
scipy                                1.14.1
wxPython                             4.0.7
```
**mavexplorer error output**
```
jake@jake:~$ MAVExplorer.py 
/home/jake/.local/lib/python3.10/site-packages/matplotlib/projections/__init__.py:63: UserWarning: Unable to import Axes3D. This may be due to multiple versions of Matplotlib being installed (e.g. as a system package and as a pip package). As a result, the 3D projection is not available.
  warnings.warn("Unable to import Axes3D. This may be due to multiple versions of "
MAV> Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jake/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/grapher.py", line 661, in show
    self.plotit(self.x, self.y, lab, colors=col, title=self.title, interactive=interactive)
  File "/home/jake/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/grapher.py", line 270, in plotit
    self.fig = pylab.figure(num=1, figsize=(12,6))
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/pyplot.py", line 1042, in figure
    manager = new_figure_manager(
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/pyplot.py", line 551, in new_figure_manager
    _warn_if_gui_out_of_main_thread()
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/pyplot.py", line 528, in _warn_if_gui_out_of_main_thread
    canvas_class = cast(type[FigureCanvasBase], _get_backend_mod().FigureCanvas)
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/pyplot.py", line 369, in _get_backend_mod
    switch_backend(rcParams._get("backend"))
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/pyplot.py", line 425, in switch_backend
    module = backend_registry.load_backend_module(newbackend)
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/backends/registry.py", line 317, in load_backend_module
    return importlib.import_module(module_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/backends/backend_wxagg.py", line 4, in <module>
    from .backend_wx import _BackendWx, _FigureCanvasWxBase
  File "/home/jake/.local/lib/python3.10/site-packages/matplotlib/backends/backend_wx.py", line 33, in <module>
    import wx.svg
ModuleNotFoundError: No module named 'wx.svg
```