ArduPilot/ardupilot_wiki #5849: Tattu Plus DroneCAN Battery

wykzx6, 2024-03-29:
We designed a smart battery based on DroneCAN, and we hope to be able to display the single cell voltage on the mission planner. Using Hexing Cube Orange, it can already display the total voltage, total current, and battery temperature.But we can not make mission planner to dispaly single cell voltage.

rmackay9, 2024-03-29:
Hi @wykzx6,

Great that you're creating a DroneCAN battery!

So I guess you can't see the voltages here?
[image]

The battery is filling in the [BatteryInfoAux message](https://github.com/dronecan/DSDL/blob/master/ardupilot/equipment/power/20004.BatteryInfoAux.uavcan) with the cell voltages?

wykzx6, 2024-03-29:
Yes, the ground station can't display the voltage of a single battery. I can be sure that it is caused by the error of DroneCan message sent by the BMS I designed, but I can't solve this error at present.

wykzx6, 2024-03-29:
> 你好@wykzx6,
> 
> 太好了，你正在创造一个DroneCAN电池！
> 
> 我猜你看不到这里的电压？ ![image](https://private-user-images.githubusercontent.com/1498098/317931894-4ad60abe-42ec-4745-857a-d3af70597545.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE2Nzg0ODcsIm5iZiI6MTcxMTY3ODE4NywicGF0aCI6Ii8xNDk4MDk4LzMxNzkzMTg5NC00YWQ2MGFiZS00MmVjLTQ3NDUtODU3YS1kM2FmNzA1OTc1NDUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMjlUMDIwOTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDM0MDNlOGQ2MjQ5OGU5OGNlM2I1N2M3MzYxNDI1YTZjMGIwNTlkYjE0YTg0ZjgxZTc0OTVkYTVlMzg0OGI3MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.65L7YskAwiQEdEEqClCGdpIUBEYacJFB4PGHWF6KLyA)
> 
> 电池正在充电[电池信息消息](https://github.com/dronecan/DSDL/blob/master/ardupilot/equipment/power/20004.BatteryInfoAux.uavcan)电池电压？
> 
> 顺便说一下，如果你给其中一个开发人员（比如我）送一块电池，我们可以帮助测试和解决问题。当然没有压力。

wykzx6, 2024-03-29:
@rmackay9 Excuse me, which country are you from?

rmackay9, 2024-03-29:
Hi @wykzx6,

I'm based in Japan.  If you'd like to email me directly it's rmackay9 at yahoo.com.

wykzx6, 2024-03-29:
I can't use yahoo, I'm from China, do you have QQ, or you can give me your email, and I'll send you an email!

元吉
***@***.***

wykzx6, 2024-03-29:
@rmackay9 My email address is 614513094@qq.com.

rmackay9, 2024-03-29:
@wykzx6,

Great, I've just sent and email. thanks!

wykzx6, 2024-03-29:
@rmackay9 ok!

rmackay9, 2024-03-30:
I've just noticed that this issue is raised in the wiki issues list.  I think this is a C++ code issue or an issue with the peripheral.

If we think it's an ArduPilot code issue then it would be best to [re-raise the issue here](https://github.com/ArduPilot/ardupilot/issues).

I'll go ahead and close this issue but I will see replies in any case, thanks!

closed 2024-03-30 by rmackay9