ArduPilot/ardupilot_wiki #5859: Holybro Kakute F7 Mini - fault in board documentation?

nostromo123, 2024-04-05:
https://ardupilot.org/plane/docs/common-holybro-kakutef7mini.html

I think, in chapter UART Mapping there is an error.

> The UARTs are marked Rn and Tn in the above pinouts. The Rn pin is the receive pin for UARTn. The Tn pin is the transmit pin for UARTn.
>     SERIAL0 -> USB
>     SERIAL1 -> UART1 (Telem1)
>     SERIAL2 -> UART2 (Telem2)
>     SERIAL3 -> UART3 (GPS)
>     SERIAL4 -> UART4
>     SERIAL6 -> UART6 (TX only unless [BRD_ALT_CONFIG](https://ardupilot.org/plane/docs/parameters.html#brd-alt-config) = 1, then RX available also)
>     SERIAL7 -> UART7 (Receive only, ESC Telemetry)

Have that board here and stumbled over this 'BRD_ALT_CONFIG'. The parameter lists (Plane 4.4.4 and 4.5.0)  I am using don't show this parameter.

Made a post about some inaccuracies in the Wiki pages and posted
https://discuss.ardupilot.org/t/found-some-inaccuracies-in-the-wiki-how-to-help/115637

But as I did not find any response there and just stumbled over this footnote, now I am here ....

> 
Questions, issues, and suggestions about this page can be raised on the [forums](https://discuss.ardupilot.org/). Issues and suggestions may be posted on the forums or the [Github Issue Tracker](https://github.com/ArduPilot/ardupilot_wiki/issues/new?title=Holybro%20Kakute%20F7%20Mini&body=[Holybro%20Kakute%20F7%20Mini](https://ardupilot.org/plane/docs/common-holybro-kakutef7mini.html)%20-%20Describe%20the%20issue/suggestion%20and%20improve%20the%20title.%20Please%20keep%20a%20link%20to%20the%20original%20article%20if%20relevant.&labels=docs).

br Kalle

Hwurzburg, 2024-04-05:
yes...its incorrect...copy paste error I bet from KakuteF7AIO....I will update that page...tnx

Hwurzburg, 2024-04-05:
fixed....will take a day to appear on server build of wiki

closed 2024-04-05 by Hwurzburg

nostromo123, 2024-04-05:
Hope, I don't bother with theses things ....