ArduPilot/ardupilot_wiki #5860: Missing barometer telemetry values in AP Plane, yaapu and ExpressLRS.

nostromo123, 2024-04-05:
In my opinion, flying a plane  the barometer values Alt and VSpd are essential. Especially with a powered glider trying to do some thermal flying. Just learned, that the combination AP Plane, yaapu and ExpressLRS does NOT support the transmission of those values.

https://discuss.ardupilot.org/t/variometer-output-on-crsf-telemetry-for-gliders/115656

In the yaapu github I found this issue, but there Alessandro is pointig over to AP. 

https://github.com/yaapu/FrskyTelemetryScript/issues/210

Therefore this issue here ......

br Kalle

Hwurzburg, 2024-04-05:
this is not a wiki issue...code issues/feature requests are made on the arrdupilot/ardupilot github...not wiki github...its for doc issues only
the osd vspeed value is baro derived, not GPS
crsf does not provide vspeed, its  a paaassthru over crsf from the autopilot
getting vaiometer to use vspeed is an opentx issue, not AP

closed 2024-04-05 by Hwurzburg

nostromo123, 2024-04-05:
Sorry, overlook that. I'll create a new one in the right place .....

regards Kalle