ArduPilot/ardupilot_wiki #5872: "Pre Arm Generator: No Backend Driver"  message

edmarquart, 2024-04-11:
I have copter 4.5.1, herelink air and ground units, Orange cube 3+, ADS in board, Mission Planner 1.3.81. Motors and ESCs not yet connected. I am getting the message “PreArm Generator: No Backend Driver”. I have not found any info on this to work towards arming. Any advice?

Hwurzburg, 2024-04-11:
not really a doc issue...support questions should be entered on discuss.ardupilot.org
but that error is due to selecting generator type that is not actually in your firmware rev....maybe you need to build a custom firmware with your desired generator using custom.ardupilot.org

closed 2024-04-11 by Hwurzburg

edmarquart, 2024-04-11:
Thank you for this information!  I'm very new and learning.  I don't remember selecting a generator type during setup, but I could have without knowing.    I'll work through discuss.ardupilot.org to figure how to build a custom firmware.  THX