ArduPilot/ardupilot_wiki #5874: Speedy Bee F4

whoim2, 2024-04-12:
https://ardupilot.org/blimp/docs/common-speedybeef4.html
https://www.speedybee.com/speedy-bee-f4-aio-flight-controller/#Parameters

Hello! I installed arducopter 4.4.4 on this FC (this version is the latest one that supports bmp280). I can't adjust the monitor battery, voltage and current. Sensors are built-in. How can I do that?
When selecting Battery Monitor 0 > Analog Voltage and Current, the message Set BATT_AMP_PERVOLT failed and set BATT_VOLT_MULT failed.
Thank you!!

Hwurzburg, 2024-04-12:
this is not a documentation issue......support questions should be raised in the discuss.ardupilot.org forums...

but did you reboot after changing the BAT_MONITOR param?
or try resetting to default params?
I have tagged the dev for that board in

closed 2024-04-12 by Hwurzburg

whoim2, 2024-04-12:
> but did you reboot

Yes. And defaults not showing battery voltage.