ArduPilot/ardupilot_wiki #5900: Replace references to APJ Tools

rmackay9, 2024-04-25:
In AP 4.5 (and higher) we've removed APJ Tools so we should update the wiki to replace references to it with the new processes.  In particular APJ tools was used to allow OEMs to easily set the default values for parameters and mark them as read-only.

Some pages I've found:

- OEM Customisations https://ardupilot.org/copter/docs/common-oem-customizations.html
- APJ Tools https://ardupilot.org/dev/docs/apjtools-intro.html

This issue comes in response to [this user support question on the Copter-4.5 forums](https://discuss.ardupilot.org/t/safety-concerns-about-lacking-complete-configuration-validataion-in-program/116817).

Hwurzburg, 2024-04-25:
done #5901

closed 2024-04-25 by Hwurzburg

rmackay9, 2024-04-25:
Thanks very much!