ArduPilot/ardupilot_wiki #5935: Non-GPS navigation should have link to how to set the EKF origin

rmackay9, 2024-05-07:
The [Non-GPS pages](https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html) show the setup for many non-gps position sensors and all of them except Beacons require the user to set the EKF origin (unless they've also got a GPS attached).  I don't see any pages which actually show the user the two options on how this can be set:

1. use the ground station to set the origin (e.g. open MP's Flight Data screen and right-mouse-button click on the map and select Set Home Here, Set EKF Origin Here.
2. use a lua script like this one https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/examples/ahrs-set-origin.lua

Ryanf55, 2024-05-07:
This would be great to document for QGC and Mavproxy too if it's easy.

Hwurzburg, 2024-05-08:
done in AP wiki

closed 2024-05-08 by Hwurzburg

rmackay9, 2024-05-08:
Thanks so much!