ArduPilot/ardupilot_wiki #5945: DJI RS2/RS3 should warn that RC yaw control does not support "follow"

rmackay9, 2024-05-08:
As mentioned in [this issue](https://github.com/ArduPilot/ardupilot/issues/27016) our DJI RS2/RS3 driver does not support RC yaw control in "follow" mode.  RC yaw control is always "lock" mode (aka "earth frame").

We should add a warning to the [DJI RS2/RS3 wiki page](https://ardupilot.org/copter/docs/common-djirs2-gimbal.html#common-djirs2-gimbal).  Something like, "warning: Yaw rate control is always "follow" mode (aka "earth frame") meaning that if the user control the gimbal's yaw rotation rate from the RC (e.g. MNT1_RC_RATE > 0) the gimbal will rotate in "lock" mode (e.g. "earth frame", e.g. will not rotate with the vehicle).  A work around is to always use yaw angle control (e.g. MNT1_RC_RATE = 0) or switch the mount to "Neutral" whenever the user wishes to re-center it.

Hwurzburg, 2024-05-08:
done

closed 2024-05-08 by Hwurzburg

rmackay9, 2024-05-08:
Thanks again!