ArduPilot/ardupilot_wiki #5953: Dev: Setup the Build Environment on Windows could mention WSL-USB-GUI

rmackay9, 2024-05-09:
It is very useful for users to be able to attach USB devices to their WSL2 development environment which is quite easy if [WSL USB Manager](https://gitlab.com/alelec/wsl-usb-gui/-/blob/main/README.md?ref_type=heads) is used.

I think we should add an optional step to the [Setup the Build Environment on Windows](https://ardupilot.org/dev/docs/building-setup-windows.html) page to optionally recommend that users download and install it from [this link](https://gitlab.com/alelec/wsl-usb-gui/-/releases).

I've been using WSL2 for years but did not know until today that WSL USB Manager existed.  Surely I'm not the only one.

Hwurzburg, 2024-05-09:
a[[ears to be a GUI for USBIPD which IS a step in Windows setup instructions already...but GUIs are more user friendly

Hwurzburg, 2024-05-09:
Done