ArduPilot/ardupilot_wiki #5976: Dev: add details on to mavlink interface camera/gimbal page

rmackay9, 2024-05-13:
AP 4.5 supports image tracking so we should add details of these two commands to the [Control a Gimbal / Camera Mount page](https://ardupilot.org/dev/docs/mavlink-gimbal-mount.html).

- [MAV_CMD_CAMERA_TRACK_POINT](https://mavlink.io/en/messages/common.html#MAV_CMD_CAMERA_TRACK_POINT)
- [MAV_CMD_CAMERA_TRACK_RECTANGLE](https://mavlink.io/en/messages/common.html#MAV_CMD_CAMERA_TRACK_RECTANGLE)

This came up on [this QGC thread](https://github.com/mavlink/qgroundcontrol/pull/11264)

Hwurzburg, 2024-05-13:
Which PR added this...need this to keep my bookkeeping straight

rmackay9, 2024-05-13:
Hi @Hwurzburg,

This is the PR that added image tracking https://github.com/ArduPilot/ardupilot/pull/23640

Hwurzburg, 2024-05-14:
okay you did the wiki addition for that PR, will you do the wiki changes to adds these in the same format?