ArduPilot/ardupilot_wiki #5983: update RTF frames

tridge, 2024-05-14:
This list is missing some frames:
https://ardupilot.org/copter/docs/common-rtf.html

Some to add:
 - https://araceuas.com/griffin/
 - https://araceuas.com/sirin/ (actually Angel)
 - https://araceuas.com/roc/
 - https://wattsinnovations.com/pages/prismlite

TOTON95, 2024-05-14:
@tridge could you please also add [3DR Quad Zero Kit](https://store.3dr.com/3dr-quad-zero-kit/) into the list?

Thanks in advance! 🙌🏼

lida2003, 2024-05-14:
- [404 can't visit, link might not work anymore][FlyDragonDroneTech - FDG30](https://www.droneassemble.com/product/vtol-uav-6-hours-endurance-with-1kg-payload-for-survey-serveillance/)
- [Secure, Password requested, so what are we looking at][SpektreWorks - Cobalt 55 E-VTOL](https://www.spektreworks.com/cobalt)
- [Secure, Password requested, so what are we looking at][SpektreWorks - Cobalt 55 G-VTOL](https://www.spektreworks.com/cobalt)
- [Secure, Password requested, so what are we looking at][SpektreWorks - Cobalt 110 G-VTOL](https://www.spektreworks.com/cobalt)