ArduPilot/ardupilot_wiki #6065: Dev: list firmware-supported MAVLink messages and commands

ES-Alexander, 2024-06-24:
Near term:
- [x] Create firmware source parser for messages and commands that each vehicle type handles and/or sends
    - ArduPilot/ardupilot#27226
- [x] Store those outputs somewhere for existing firmwares
    - ArduPilot/ParameterRepository#7
- [ ] Add per-vehicle dropdown version selection support, [like the parameter lists](https://github.com/ArduPilot/ardupilot_wiki/blob/master/build_parameters.py#L338)
- [ ] Include in the wiki
    - Pull the files in during generation
    - Vaguely relevant to #1884, although that's focused more on higher level overviews
    - Also somewhat relevant to #74, although the current auto-parsing is only for listing message names and the source file they're handled in / sent from

Longer term:
- [ ] Set up automatic generation per vehicle, per version
    - May be done via autotest, [like for parameters](https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/autotest.py#L194)
        - Moving the script to autotest could be useful for notifying of changes in MAVLink support on PRs, but is probably too much effort for now
    - Will likely get done externally, in what is currently [the parameters repository](https://github.com/ArduPilot/ParameterRepository/blob/main/scripts/run_param_parse.py#L132)
        - Extra attractive because the script requires Python &geq; 3.11, which will be easier to specify there than forcing autotest to update
        - **WIP**: ArduPilot/ParameterRepository#8
- [ ] Account for / mention preprocessor directives, where relevant
    - e.g. add a "notes" column, and include things like "message only included if firmware compiled with `XXX` flag"
- [ ] Include supported fields, and any definition/range specifics for the selected vehicle
    - For automatic parsing would likely require an explicit comment syntax for within the firmware source

zaccesss, 2026-11-05:
@ES-Alexander I have opened #8234 for the per-vehicle version selector and @Hwurzburg has already reviewed it. I would like to help with the parser items next, starting with "Account for / mention preprocessor directives, where relevant". Before I start, how would you like it done?

- Should the notes come from [`Tools/scripts/mavlink_parse.py`](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/mavlink_parse.py) itself, by recording the `#if` guards around each handler and sender? The other way would be a hand-maintained list.
- For the output, is a "Notes" column in the existing tables what you had in mind, for example "only when built with `HAL_MOUNT_ENABLED`" (used around the gimbal handlers in [`GCS_Common.cpp`](https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp))?
- Since the files are now generated automatically (ArduPilot/ParameterRepository#8), is the parser in the ardupilot repo still the right place for this change?

I can look at "supported fields" after that. As the issue says, that one needs a comment syntax in the firmware source, so I would rather follow your lead on the format.

ES-Alexander, 2026-11-09:
Hi @zaccesss, thanks for your interest in helping with this :-)

> Should the notes come from [Tools/scripts/mavlink_parse.py](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/mavlink_parse.py) itself, by recording the #if guards around each handler and sender? The other way would be a hand-maintained list.

Yes, this should be automated by detection of `#if` guards from the code, and included in the output as a column named "part of" or "requires" or "feature" or similar, to indicate which broader feature/library the message is part of / which compilation flag it requires. That said, there may be some complexity around determining that effectively, because a message may be received/sent in multiple places, in which case ideally independent requirements could be independently linked to (or at least listed), but that may make the generation script overly complex. It would be nice if we can at least determine the most general requirement (e.g. if a message is handled unguarded and also separately with a guard, then the minimal requirement would be nothing, but we could alternatively specify it's part of "general / `feature`").

While discussing ideals, it would be valuable to add an extra output column for "included by default?", but there's complexity there in differentiating between 
1. "yes" (including when [`build_options.py`](https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/build_options.py) is applied, as part of building with the custom build server), or
2. "in official releases" (tracking `#if {FEATURE}_ENABLED` guards and their defaults (typically defined in `_config.h` files), including multiple levels (e.g. the `NMEA_UNICORE` feature set is [included when NMEA GPS is included](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_config.h#L73), which follows a chain up to GPS in general being included by default), or 
3. "in ArduSub" (or similar, for features guarded by vehicle type checks), or 
4. "with enough flash" (or some better name, for features disabled in the `https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_HAL_ChibiOS/hwdef/include/minimize_*.inc` files, or in several individual board hwdef files).

> Since the files are now generated automatically ..., is the parser in the ardupilot repo still the right place for this change?

Yes, the automatic generation in the parameter repo regularly does updated generations using the latest version of the parser in the ArduPilot repo.

> I can look at "supported fields" after that. As the issue says, that one needs a comment syntax in the firmware source, so I would rather follow your lead on the format.

There's a bit of nuance there - a "minimal" solution in terms of codebase changes may just check for which fields are even ingested/used (since some of them get ignored, especially when new fields are added and the code hasn't caught up), but that may not be straightforward if the handling is done by functions defined elsewhere in the codebase. The parsing of supported fields would be much more straightforward if we had a comment directive syntax for it in the source, but that may be challenging to get agreement on, given it adds maintenance load and feature support is not always binary, especially if a field can be used for multiple purposes...

zaccesss, 2026-11-12:
@ES-Alexander thanks for the detail, that gives me a clear direction. This is how I plan to do it, so you can correct me before I start:

- The guards are detected automatically in `Tools/scripts/mavlink_parse.py`, with no hand-maintained list.
- The output gains one column. I will call it "Requires" unless you prefer "Part of" or "Feature".
- When a message is handled or sent in more than one place, I will show the most general requirement first. So a message that is handled unguarded in one place and guarded in another shows no requirement. If listing each guard separately turns out to be simple, I will list them. If it makes the script much more complex, I will stop at the most general one and say so in the PR.
- I will leave the "included by default?" column for a follow-up. The four cases you describe (build options, release defaults through the `_config.h` chain, vehicle checks and the `minimize_*.inc` files) are a separate piece of work.

I will open the PR in ArduPilot/ardupilot and link it here. The Stream Groups section that is hand-added to the wiki pages (see #8234) could go into the parser in the same piece of work, if you agree.

On supported fields, understood. I will leave that until there is agreement on a comment syntax.

ES-Alexander, 2026-11-12:
That sounds fine to me - thanks :-)

Leave out the Stream Groups section, per the PR discussion.

zaccesss, 2026-11-13:
> Leave out the Stream Groups section, per the PR discussion.

Done. The Requires column is in ArduPilot/ardupilot#35148 on its own, with no Stream Groups changes.

On the discussion in #8234:

> As I understand it the firmware stopped supporting stream groups in 4.7

I came across this while working on the column. It looks like they are still present in 4.7 and later. They seem to have moved from each vehicle's own file to a shared file, `libraries/GCS_MAVLink/GCS_MAVLink_Parameters.cpp`, in ArduPilot/ardupilot#29617. The parser still looks for the old file, which may be why it finds nothing there. Please correct me if I have misread it.

So I have also opened two drafts, in case they are useful:

- ArduPilot/ardupilot#35149 builds on the first one and lets the parser find stream groups in every version.
- ArduPilot/ParameterRepository#22 is the matching one-line change.

Both are drafts. Please close them if you would prefer to leave Stream Groups out. The first pull request stands alone either way.