ArduPilot/ardupilot_wiki #6155: Support virtual environment for python

Ryanf55, 2024-08-01:
# Request

Can we have the wiki setup work in a python virtual environment?

https://github.com/ArduPilot/ardupilot_wiki/issues/358#issuecomment-2263025787

I'm having issues with the `flake8` install hosing my ROS 2 install which has flake8 installed as a debian. 

# Expected behavior

```
python3 -m venv .venv
source .venv/bin/activate
./Spinxsetup.sh
python3 update.py --site dev
```

# Actual  behavior

```
python3 -m venv .venv
source .venv/bin/activate
./Spinxsetup.sh
```
```
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
```

Hwurzburg, 2024-08-01:
not sure who can work on this

Ryanf55, 2024-08-01:
At least the issue is filed. Maybe someone will have time to solve it. I manually installed things instead of using the script so I can work in the wiki.

khancyr, 2024-08-01:
Ok that is just a matter of removing the --user when inside a venv. Patch should be simple, can do it tonight