ArduPilot/ardupilot_wiki #6171: Splined WPs removed/not working

billjow, 2024-08-08:
I have Arduplane 4.5.5 installed on a F405 flight controller in a Fimi Manta VTOL. In an effort to smooth out a mission, I tried to use Spline WPs. If I load an existing mission into the Plan section of Mission Planner, the waypoints show up as Unknown and cannot be dragged around (see attached screeenshot [image]). If I am creating a new mission, I can insert a Spline WP after a numbered waypoint, but have to manually edit the coords for it to show. It only shows up as a Waypoint in Mission Planner. Please could the previous Spline WP functionality be reinstated. Thanks!
[image]

Hwurzburg, 2024-08-08:
this is not a wiki issue, its a support question(should be posted on discuss.ardupilot.org or discord plane-help)....
but fyi...plane(and quadplane) has never supported spline waypoints (or Scurves)....MP creates missions that are vehicle agnostic...but the vehicle code determines what MAV commands are implemented for that vehicle

closed 2024-08-08 by Hwurzburg

billjow, 2024-08-08:
Understood and thanks for the information. Kind Regards, Bill Sent from my iPhone