ArduPilot/ardupilot_wiki #6201: Incorrect april tag bundle labelling

chewdonovan, 2024-08-20:
[Indoor autonomous flight with ArduCopter, ROS and AprilTag Detection](https://ardupilot.org/dev/docs/ros-apriltag-detection.html)

[image]

The provided april tag bundle is labeled from 0-5 and 7-12, where it should be 0-11.

As can be seen the tag labelled "7" is actually the number 6 tag.

[image]

@thomas-matthews-chironix

rmackay9, 2024-08-20:
Indeed, the number has missed "6"!  Thanks for the report!

Hwurzburg, 2024-08-20:
I dont know of anyone that can fix this, to assign it to

chewdonovan, 2024-08-20:
@Hwurzburg

Just in case you find someone

[image]

Hwurzburg, 2024-08-20:
I can replace with those...thanks

Hwurzburg, 2024-08-20:
replaced https://github.com/ArduPilot/ardupilot_wiki/pull/6203

closed 2024-08-20 by Hwurzburg