ArduPilot/ardupilot_wiki #6223: Mission Planner Overview

zhangyihangceshi1hao, 2024-08-29:
Void SvgNet.SvgGdi.GdiGraphics..ctor(System.Drawing.Graphics
[image]
[image]

Hwurzburg, 2024-08-30:
not reader of Chinese,I have no  idea what this issue is about

rmackay9, 2024-08-30:
Through the wonders of ChatGPT, it seems to be a mission planner compile issue.  @zhangyihangceshi1hao could you please raise this over in the [Mission Planner's issues list](https://github.com/ArduPilot/MissionPlanner)?  

Main Error Message: To use the designer, the following errors must be resolved:

2 errors

Could not find type: "Void SvgNet.SvgGdi.GdiGraphics..ctor(System.Drawing.Graphics)".

Error Details:

This error occurred in the following instance (1): Hidden Call Stack
at MissionPlanner.Controls.HUD..ctor()

Relevant Error Help: No help associated with this error was found. Check Windows Desktop Development Troubleshooting List.

Related Error Forum Threads: Search MSDN forums for threads related to this error.

Variable 'hud1' is declared but never assigned.

closed 2024-08-30 by rmackay9