ArduPilot/ardupilot_wiki #6245: Hobbywing DroneCAN ESC

YF-CHA, 2024-09-05:
[Hobbywing DroneCAN ESC](https://ardupilot.org/copter/docs/common-hobbywing-dronecan-esc.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

After dronecan was used to drive the motor, click on the motor test, the dronecan communication content did not change, and the motor test plate of mission planner could not issue the throttle command, and the pwm was used before to drive the motor.

rmackay9, 2024-09-05:
Hi @YF-CHA,

I think the motors can take a while to spin up.  Could you try setting the MP motor test timeout to more like 5 seconds?

rmackay9, 2024-09-05:
By the way, this is really a support issue so it doesn't belong on the wiki to-do list so I will close this.  Sorry to be strict but if we're not then our to-do list turns into another [support forum](https://discuss.ardupilot.org/c/arducopter/copter-45/174).

closed 2024-09-05 by rmackay9