ArduPilot/ardupilot_wiki #6273: Serial Passthrough documentation needs correction and improvement

brad112358, 2024-09-12:
https://ardupilot.org/copter/docs/common-serial-passthrough.html incorrectly states that SERIALx_PROTOCOL should be set to -1("None").  

This would seem to be bad advice since this disables Rx and Tx pins at the driver level.  Instead, we should note that when SERIAL_PASS2 is set to non-zero, both serial ports are locked, temporarily blocking their normal use.

In addition, it would be nice to add a note that documents the special handling of USB ACM ports which will automatically set the baud rate and other settings of the SERIAL_PASS2 port to match the USB ACM host serial settings.  This means, for example, that the user does not need to modify their normal settings for SERIALx_BAUD in the common use case where SERIAL_PASS1 is a USB port.

It might also be helpful to note at the bottom that setting SERIAL_PASSTIMO to 0 avoids the passthrough timeout entirely, allowing passthrough operation until the Flight Controller is rebooted.  In fact, recommending this rather than 30 to 60 seconds may help avoid user support issues caused by the timeout expiring before the host serial application is started by the user.

peterbarker, 2024-09-13:
@brad112358 did you want to put a PR forward with your suggested changes?

brad112358, 2024-09-13:
> @brad112358 did you want to put a PR forward with your suggested changes?

Sure, I can do that.  Do all three changes I proposed sound OK?

Hwurzburg, 2024-09-13:
I thought it needed some other updates so I just pushed a PR rather than typing a lot of words here (currently one armed)

closed 2024-09-14 by Hwurzburg